pimoroni/automation-hat #5: .toggle() not working for some features

mnelsoneorm, 2016-10-09:
I can .toggle() relays, but toggle does nothing for comms, warn, outputs.  on() and off() work fine.  No error message.  Just nothing happens.  Does it not know current "state" to toggle from/to?

Gadgetoid, 2016-10-10:
I'm looking into this. For `comms`, `warn` etc it was just awful broken code (I need to find a way to add test coverage to code that interacts with hardware!).

Gadgetoid, 2016-10-10:
This should be fixed by 8643d543513b650b872180830415b4960a1955ad and 90b45b56e641e9fb19e6c1381380e71ce99ad04c which I'll push out in Version 0.0.2.

bfqrst, 2016-11-01:
On a similar note, the LEDs of the ADC are supposed to be somewhat voltage sensitive from what I've learned from the corresponding Bilge Tank episode. Now those guys aren't lighting up at all in my case. They're not defective, I can tell you that much, because the on() or write() function actually switches them on (or off). That's how I found out that toggle() doesn't work... Maybe you guys can look into this as well.

I'm happy to transfer this into a dedicated GitHub issue if it helps?!

Cheers
Ralph

Gadgetoid, 2016-11-08:
Hopefully 0.0.2 has sorted this out, but if not let me know and it'll make 0.0.3.

closed 2016-11-08 by Gadgetoid