pimoroni/automation-hat #10: Problems with Raspbian

stefancrain, 2017-06-23:
Hi all, 

I'm hoping you can update this script to work with the most recent release of Raspbian / Jesse. 

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.28-v7+ #998 SMP Mon May 15 16:55:39 BST 2017 armv7l GNU/Linux

pi@raspberrypi:~ $ lsb_release -a
LSB Version:	core-2.0-armhf:core-2.0-noarch:core-3.0-armhf:core-3.0-noarch:core-3.1-armhf:core-3.1-noarch:core-3.2-armhf:core-3.2-noarch:core-4.0-armhf:core-4.0-noarch:core-4.1-armhf:core-4.1-noarch:security-4.0-armhf:security-4.0-noarch:security-4.1-armhf:security-4.1-noarch
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
```
I'm seeing some errors on a fresh install, I'm sure I wont be the only one. 
```
pi@raspberrypi:~ $ curl https://get.pimoroni.com/automationhat | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 38504  100 38504    0     0  42453      0 --:--:-- --:--:-- --:--:-- 42452

This script will install everything needed to use
Automation HAT

Always be careful when running scripts and commands copied
from the internet. Ensure they are from a trusted source.

If you want to see what this script does before running it,
you should run: 'curl https://get.pimoroni.com/automationhat'

Note: Automation HAT requires I2C communication

Do you wish to continue? [y/N] y

Checking environment...
Updating apt indexes...
.....................
Installing python-pip...
..................Python 3 is not installed. Would like to install it? [y/N] ...y
Installing python3-pip...
.........E: Problem renaming the file /var/cache/apt/pkgcache.bin.kfNoOs to /var/cache/apt/pkgcache.bin - rename (2: No such file or directory)
W: You may want to run apt-get update to correct these problems
Apt failed to install python3-pip!
Falling back on pypi...

Checking hardware requirements...

Checking for packages required for GPIO control...
..................bash: line 708: list: command not found
Unable to install RPi.GPIO for python 3!

I2C must be enabled for Automation HAT to work
...I2C is now enabled

Checking packages required by I2C interface...
............^C
```

I canceled, hoping I could document this and move on. 

```
pi@raspberrypi:~ $ curl https://get.pimoroni.com/automationhat | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 38504  100 38504    0     0  50072      0 --:--:-- --:--:-- --:--:-- 50070

This script will install everything needed to use
Automation HAT

Always be careful when running scripts and commands copied
from the internet. Ensure they are from a trusted source.

If you want to see what this script does before running it,
you should run: 'curl https://get.pimoroni.com/automationhat'

Note: Automation HAT requires I2C communication

Do you wish to continue? [y/N] y

Checking environment...
Updating apt indexes...
.....................
Installing python3-pip...
.........
Checking hardware requirements...

Checking for packages required for GPIO control...
......DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
......RPi.GPIO installed and up-to-date

I2C already enabled

Checking packages required by I2C interface...
.....................DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
smbus installed and up-to-date

Automation HAT comes with examples and documentation that you may wish to install.
Performing a full install will ensure those resources are installed,
along with all required dependencies. It may however take a while!

Do you wish to perform a full install? [y/N] y

Checking for dependencies...
python-sn3218 is required
Installing python-sn3218...
python3-sn3218 is required
Installing python3-sn3218...

Installing python-automationhat...
install ok installed
Installing python3-automationhat...
install ok installed

Checking for additional software...
Installing git...

Downloading examples and documentation...
Resources for your Automation HAT were copied to
/home/pi/Pimoroni/automationhat

All done. Enjoy your Automation HAT!
```

From there I can attach the hat and run an example.

Gadgetoid, 2017-06-26:
As far as I can tell, your first error was a one-off hiccup on Apt's part. I couldn't replicate it.

The `DEPRECATION` warning output by `pip list` is interesting, though. What do you see if you run:

`pip --version` and `pip3 --version`

Gadgetoid, 2017-11-20:
Any success with this since? I'm going to close this issue, but you're welcome to re-open if you're still having issues.

closed 2017-11-20 by Gadgetoid

dh-harald, 2017-11-24:
Fresh (/updated) raspbian stretch install:
`$ curl https://get.pimoroni.com/microdotphat | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 38511  100 38511    0     0  80315      0 --:--:-- --:--:-- --:--:-- 80398

This script will install everything needed to use
Micro Dot pHAT

Always be careful when running scripts and commands copied
from the internet. Ensure they are from a trusted source.

If you want to see what this script does before running it,
you should run: 'curl https://get.pimoroni.com/microdotphat'

Support for your operating system is experimental. Please visit
forums.pimoroni.com if you experience issues with this product.

Note: Micro Dot pHAT requires I2C communication

Do you wish to continue? [y/N] y

Checking environment...
Updating apt indexes...
...............
......Installing python-pip...
Extracting templates from packages: 100%
......Installing python3-pip...
.........
Checking hardware requirements...

I2C must be enabled for Micro Dot pHAT to work
...No passwd entry for user 'pi'
...No passwd entry for user 'pi'
I2C is now enabled

Checking packages required by I2C interface...
.................................DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Unable to install smbus for python 2!
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Unable to install smbus for python 3!

Micro Dot pHAT comes with examples and documentation that you may wish to install.
Performing a full install will ensure those resources are installed,
along with all required dependencies. It may however take a while!

Do you wish to perform a full install? [y/N] y

Checking for dependencies...
python-numpy is required
Installing python-numpy...
python3-numpy is required
Installing python3-numpy...

Installing python-microdotphat...
install ok installed
Installing python3-microdotphat...
install ok installed

Checking for additional software...
Installing git...

Downloading examples and documentation...
Resources for your Micro Dot pHAT were copied to
/home/xxx/Pimoroni/microdotphat

All done. Enjoy your Micro Dot pHAT!

Some packages could not be installed, review the output for details!

Support for your operating system is experimental. Please visit
forums.pimoroni.com if you experience issues with this product.
$ pip --version
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)
$ pip3 --version
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.5)
$ dpkg -l | grep python-smbus
ii  python-smbus:armhf              3.1.2-3                      armhf        Python bindings for Linux SMBus access through i2c-dev
$ dpkg -l | grep python3-smbus
ii  python3-smbus:armhf             3.1.2-3                      armhf        Python 3 bindings for Linux SMBus access through i2c-dev`