pimoroni/automation-hat #11: Receiving 'Automation pHAT detected...' message with no hat connected

stefancrain, 2017-06-23:
With a fresh install of raspbianI tried to use [this example file](https://github.com/pimoroni/automation-hat/blob/master/examples/relay.py). I had no hat attached and had never attached a hat. I still see that it detected one and promptly failed. 
```
pi@raspberrypi:~ $ python3 ./example.py
Automation pHAT detected... 
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3/dist-packages/automationhat/pins.py", line 37, in run
    if self.todo() is False:
  File "/usr/lib/python3/dist-packages/automationhat/__init__.py", line 293, in _update_adc
    analog._update()
  File "/usr/lib/python3/dist-packages/automationhat/pins.py", line 92, in handler
    return self._do(name, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/automationhat/pins.py", line 111, in _do
    _results[node] = handler(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/automationhat/__init__.py", line 122, in _update
    self.value = ads1015.read(self.channel)
  File "/usr/lib/python3/dist-packages/automationhat/ads1015.py", line 42, in read
    self.i2c_bus.write_i2c_block_data(self.addr, REG_CFG, [(config >> 8) & 0xFF, config & 0xFF])
OSError: [Errno 121] Remote I/O error

^CTraceback (most recent call last):
  File "./example.py", line 16, in <module>
    time.sleep(0.1)
KeyboardInterrupt
```
Seems like thats an error with detecting that the hat is connected.

Gadgetoid, 2017-06-26:
There's no real detection happening, the library is just attempting to differentiate between pHAT and HAT by checking for the presence/absence of certain i2c devices (the sn3218 in this case). It's a pretty naive check and is definitely not bulletproof. Yes, the library will fail if you don't have a HAT or pHAT connected.

RogueM, 2017-06-26:
to add to this though - I am aware that there is a gap on the logic when no HAT/pHAT is attached, it should claim a pHAT is attached.

I'll look at and try to improve when I have a minute but apart from obviously not claiming a pHAT is attached when none are that is not going to magically make things work when there is no hardware attached ;-)

stefancrain, 2017-06-26:
@RogueM Thats exactly what I was looking for. I was trying to debug a hardware failure last month and it was very confusing to see that my Hat had changed into a pHat.