pimoroni/automation-hat #14: can't install on python3.6

harper357, 2017-11-26:
I've tried both manually compiled python and linuxbrewed python and neither one seems to work when i try to pip3 install automationhat. I think the problem comes from the back that smbus isn't installed and I can't find a way to install that.

Gadgetoid, 2017-11-29:
Presently we only support Raspbian Stretch and Raspbian Jessie and their associated Python versions with any sort of regularity- this is because there's simply not enough time to go far beyond this.

It sounds like you need to find a version of SMBus for Python 3.6, or compile your own.

closed 2017-11-29 by Gadgetoid

gvier, 2018-08-27:
I had the same problem trying to use automationhat with python 3.6 on piCore 9.0.3.
I solved it by using the smbus2 python library and doing a few code changes in both automationhat and sn3218 libraries:
* replacing _smbus_ with _smbus2_
* _read_i2c_block_data_ now takes a third parameter for the length of bytes to read.

Hope this helps.