pimoroni/automation-hat #17: No ADC detected, check your connections

EdoB, 2018-04-20:
Hi,
after a fresh install of Raspbian Stretch (kernel 4.14) I'm unable to make the AutomationHat work. It seems that the RPi can't pickup the board.

The installation with `sudo curl get.pimoroni.com/automationhat | bash` returns no error, but when I try to use the library with the Python console it returns at first a RuntimeError, and then an IOError:
```
pi@raspberrypi:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2
>>> import automationhat
>>> automationhat.analog.one.read()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/automationhat/__init__.py", line 119, in read
    self._update()
  File "/usr/lib/python2.7/dist-packages/automationhat/__init__.py", line 123, in _update
    self.setup()
  File "/usr/lib/python2.7/dist-packages/automationhat/__init__.py", line 106, in setup
    setup()
  File "/usr/lib/python2.7/dist-packages/automationhat/__init__.py", line 357, in setup
    raise RuntimeError("No ADC detected, check your connections")
RuntimeError: No ADC detected, check your connections
>>> automationhat.analog.one.read()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/automationhat/__init__.py", line 119, in read
    self._update()
  File "/usr/lib/python2.7/dist-packages/automationhat/__init__.py", line 124, in _update
    self.value = _ads1015.read(self.channel)
  File "/usr/lib/python2.7/dist-packages/automationhat/ads1015.py", line 42, in read
    self.i2c_bus.write_i2c_block_data(self.addr, REG_CFG, [(config >> 8) & 0xFF, config & 0xFF])
IOError: [Errno 121] Remote I/O error
>>> automationhat.input.one.read()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/automationhat/__init__.py", line 190, in read
    value = Pin.read(self)
  File "/usr/lib/python2.7/dist-packages/automationhat/__init__.py", line 143, in read
    self.setup()
  File "/usr/lib/python2.7/dist-packages/automationhat/__init__.py", line 180, in setup
    setup()
  File "/usr/lib/python2.7/dist-packages/automationhat/__init__.py", line 357, in setup
    raise RuntimeError("No ADC detected, check your connections")
RuntimeError: No ADC detected, check your connections
>>> automationhat.input.one.read()
0                              <--- HERE IT RETURNS THE VALUE
>>> automationhat.input.one.read()
1                              <--- AND IT READS THE PROPER CHANGE  
```

With the relay commands is very odd, it says that the board is a pHAT:
```
>>> automationhat.relay.one.on()           <--- The relay ticks but there is no change between NC and NO
>>> automationhat.relay.two.on()
/usr/lib/python2.7/dist-packages/automationhat/__init__.py:276: UserWarning: Relay 'two' is not supported on Automation pHAT
  warnings.warn("Relay '{}' is not supported on Automation pHAT".format(self.name))
```

I then checked for i2c connectivity and in fact the board is not detected: 
```
$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

I updated everything I could think of but the outcome is the same.
```
sudo apt-get update && sudo apt-get upgrade -y
sudo rpi-update
```

Do you know what can I do next?

Gadgetoid, 2018-04-23:
What version of the Pi are you using, and do you have any other i2c HATs/pHATs to test on it?

EdoB, 2018-04-23:
It is a RPi3.
I was able to get my hands on a second AutomationHat and the behavior is the same as before. Perhaps something has happened with my Pi even if I just fresh install it. I will try a new one and get back to you.

EdoB, 2018-05-02:
With a new RPi3 everything works as expected.

closed 2018-05-02 by EdoB