pimoroni/automation-hat #18: Inputs lack handlers that are present on explorer-hat

shortbloke, 2018-06-12:
I'm in the process of implementing node-red support for the Automation-Hat. Looking at the input side, this python library only supports reading the inputs. Is there a reason that the automation-hat library doesn't support the analog changed event, and the digital input changed, high and low events.? 
Interestingly the input handler also include debouncing the digital inputs

Any chance this could be added, or is there a reason for this difference? Thanks

Gadgetoid, 2018-06-13:
I deliberately left out a lot of the more edge-case functionality, since Automation HAT is targeted at a slightly more advanced userbase and all the threading nonsense would probably cause more problems than it would solve (see some of the historic problems with ADC reads). I'm not dead set against the idea, though, since these features can be useful even to power users. (if not arguably more useful)

shortbloke, 2018-06-13:
I'll go and take a look at historic issues with ADC on both AutomationHAT and ExplorerHAT projects. I know my quick attempts last night to port the Analog input changed functionality over led to very erratic ADC with the lights flickering all over the place even when the input was connected to 0v or 5v. I didn't get the time to determine if this was being caused by the node-red side or the python library. I'd need to take a more systematic approach I think.

Gadgetoid, 2020-03-10:
Closing this thread since it seems to have gone stale. Feel free to re-open if you're still interested! Thank you.

closed 2020-03-10 by Gadgetoid