pimoroni/automation-hat #24: ADC4 not providing sensible readings

shortbloke, 2019-10-28:
I was seeking to implement ADC4 support in my Node-Red module for the Automation HAT and have the groundwork in place. However, it appears that the library is not providing sensible readings for the 3.3V ADC4.

- Connecting ADC4 to GND, gives readings of: 0.58 - 0.59
- Connecting ADC4 to 3v3, gives readings of: 0.63 - 0.65

I only have a single Automation HAT to test against, so can't rule out hardware, though I think it's unlikely as I'd never used those headers until today and I seem to recall they always gave low/odd readings which was why I excluded it from my initial implementation.

Environment:
- Raspberry Pi 3 Model B
- Raspbian Buster (Linux pi3-red 4.19.75-v7+ #1270 SMP)
- Python 3.7.3

shortbloke, 2019-10-29:
Based on response to a query on twitter this issue might just be me and an issue with my HAT. Though would nice to get confirmation.

Gadgetoid, 2019-11-01:
It's possible. Do you know what version of the Automation HAT library you have installed? There have been fixes relating to the ADC which might be relevant if it's an old install.

shortbloke, 2019-11-01:
I used the latest version directly from source here. I then went and tested using other ads1015 libraries with the same result, so suspect hardware. I've purchased another HAT which should arrive today and I'll test and confirm this weekend.

Gadgetoid, 2019-11-01:
Yep, that sounds like hardware. Good luck!

shortbloke, 2019-11-01:
Confirmed, ADC4 on new Automation HAT works fine. Closing

closed 2019-11-01 by shortbloke