pimoroni/automation-hat #31: Automation Hat Mini LCD from other languages

BryanCrotaz, 2020-06-04:
You provide a python interface, but no C library etc. I'm trying to interface from C# via wiringPi but cannot get images to appear. Could you provide a document listing all the control commands and GPIO pins (eg CS) that are relevant, and show the waveforms required to make it work?

BryanCrotaz, 2020-06-04:
Can I just leave the chipselect pin low the whole time, or does it need to be cycled?

Gadgetoid, 2020-07-31:
I don't have the time or expertise to support anything beyond Python at the moment- too many products, and just one me.

Details of the pins are here- https://pinout.xyz/pinout/automation_hat_mini

The display is the ST7735, the Python library is here which can be a useful starting point - https://github.com/pimoroni/ST7735-python

Other details are available in the display datasheet- https://www.displayfuture.com/Display/datasheet/controller/ST7735.pdf

Looks liek there might already be a C library out there somewhere, I found this at least: https://github.com/Matiasus/ST7735

BryanCrotaz, 2020-07-31:
Could you add those notes to the front page of the docs?

Gadgetoid, 2020-07-31:
I don't see any reason why not. The pinout is already there but a datasheet link and some examples of alternate libraries wouldn't go amiss.

thirdr, 2025-01-07:
Closed as stale

closed 2025-01-07 by thirdr