pimoroni/automation-hat #34: Not an issues just a question Rasberry pi and PIM213 interface

stmark707, 2020-07-15:
I am looking into this hat due to its capabilities but one of the things I am unsure about is it's way of connecting to the GPIO pins. I cannot seem to find a photo that shows how it connects to the I2C pins, voltage pins, and GPIO pins does it sit on top of the pins like some other relay boards? Can someone show me their setup? it is very important to me because this will be used in an industrial setting. Any photos or clarification on how it connects to the raspberry pi will be much appreciated and helpful.

Gadgetoid, 2020-07-17:
Like all HATs, it plugs securely onto the Pi's main header- it uses a 40pin (2x20) socket header that connects to all 40 of the pins on the Pi.

[image]

In situations where you need to ensure good mechanical coupling, you can also securely bolt the HAT to the Raspberry Pi through the mounting holes. As illustrated here by Cluster HAT:

[image]

I appreciate these might not illustrate exactly what you wanted to see, so I'll try to grab a low-angle photo of the Automation HAT Mini I've got set up.

Gadgetoid, 2020-07-17:
Here’s a side-on shot of the headers connecting between HAT and Pi in my setup:

[image]

closed 2022-02-14 by Gadgetoid