pimoroni/automation-hat #41: Problems with installation - Module ST7735 not installed and SPI not enabled

JacksonBailey, 2021-04-11:
## Quick summary

Going through the install tutorial and script a few things are missed

1. Module `ST7735` is not installed
2. SPI interface is not enabled or mentioned that it needs to be

## More detailed

Hi, after falling the instructions from [the tutorial on pimoroni.com](https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-automation-hat-and-phat) I get the error below when trying to run the example.

```
pi@pizero:~/Pimoroni/automationhat/examples/hat-mini $ python3 output.py
Traceback (most recent call last):
  File "output.py", line 17, in <module>
    import ST7735 as ST7735
ModuleNotFoundError: No module named 'ST7735'
```

I'm not great with Python but it seems pretty straight forward that a package called "st7735" might be missing, so I punched it into Google and found the [`ST7735` module](https://pypi.org/project/ST7735/). I get a small error below mentioning that `python-imaging` isn't available, I will open an issue on their code base too to be complete but I'm mentioning it here since it is part of the install process I needed to follow to get everything working. (However, in the end I did get the example working even without `python-imaging` or manually installing `python-pil` as it suggests may be needed.)

```
pi@pizero:~/Pimoroni/automationhat/examples/hat-mini $ sudo apt-get update && sudo apt-get install python-rpi.gpio python-spidev python-pip python-imaging python-numpy
[sudo] password for pi:
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.0 MB in 32s (402 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package python-imaging is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python-pil

E: Package 'python-imaging' has no installation candidate
```

The final error I encountered was this:

```
pi@pizero:~/Pimoroni/automationhat/examples/hat-mini $ python3 output.py
output.py

This Automation HAT Mini example toggles and displays the
status of the three 24V-tolerant digital outputs.

Press CTRL+C to exit.

Traceback (most recent call last):
  File "output.py", line 52, in <module>
    spi_speed_hz=4000000
  File "/usr/local/lib/python3.7/dist-packages/ST7735/__init__.py", line 156, in __init__
    self._spi = spidev.SpiDev(port, cs)
FileNotFoundError: [Errno 2] No such file or directory
```

I wasn't sure what this was so I just enabled the SPI through `raspi-config`'s interface options and now the example works just fine.

Gadgetoid, 2021-04-12:
Looks like ST7735 has been correctly added as a dependency here for a while, but the underlying problem must be the lack of "PIL" or `python-imaging` and ST7735 appears to have some *quite* old installation instructions. I'll update those.

JacksonBailey, 2021-04-12:
@Gadgetoid Just to be clear the module `ST7735` was not added by running `curl https://get.pimoroni.com/automationhat | bash`, I had to manually add it myself. Just specifying because looks like I didn't make this super clear in the first post. So even though it may be a dependency listed here `curl https://get.pimoroni.com/automationhat | bash` didn't install it.

ahepp, 2021-07-18:
I had numerous issues along these lines when I tried to run the analog example for the hat-mini on the pi 0-w. I resolved the issues by:
- `pip3 install ST7735`
- `sudo apt install python3-numpy` (note that `pip3 install numpy`  led to additional errors)
- pip3 install fonts font-roboto
- enabling SPI with `raspi-config`
- changing the working directory to Pimoroni/automationhat/examples/hat-mini

grossadamm, 2023-11-11:
@ahepp thank you!

closed 2024-05-07 by helgibbons