pimoroni/automation-hat #47: Analog input weirdness with new ADS1115 chip.

wonderly2, 2022-01-12:
I have a Hat with the new ADS1115 chip. Out of the box, the analog inputs were all screwed up using the mfg supplied library.

Tested analog inputs utilizing onboard 5V and examples/hat/inputs.py.
When no inputs are not connected, 'one' reads 4.42V and led 1 is on.
When 5V connected to input 1: 'two' reads 5.16 and led 1 & 2 are on.
When 5V connected to input 2: 'three' reads 5.17 and led 1 & 3 are on.
When 5V connected to input 3: 'four' reads 0.66 and led 1 is on.

Found solution with two edits to ads1015.py described here:
https://forums.pimoroni.com/t/automation-hat-analogue-in-spurious-reading/10749/7

Gadgetoid, 2022-01-13:
Thank you for delving into this- it looks like your suggested fix, do the detect at setup, is what we're already doing in the standalone version of the ADS1x15 library... more or less. Would be prudent to throw away the duplicate code here and migrate over.

wonderly2, 2022-01-13:
Hey, I'm not really sure what you meant there, as far as what the standalone version is, or where there is duplicate code. But I think some code needs to be added. Whatever python script I downloaded using the pimoroni bash setup script appeared to talk about detecting which chip is installed, but didn't include the samples per second map for the new chip. 
Figured you need something along the lines of:
```
if ads1015_present:
  SAMPLES_PER_SECOND_MAP = {....}
else:
  SAMPLES_PER_SECOND_MAP = {....} #ads1115 map
```
You probably understood that tho.

Gadgetoid, 2022-01-18:
This should hopefully be fixed in version 0.4.0.

I meant that we have a standalone library- here: https://github.com/pimoroni/ads1015-python/ - which does not have this issue, and should be used by this library instead of its own in-library copy of the ADS1015 code.

closed 2022-02-14 by Gadgetoid