pimoroni/automation-hat #50: automation mini hat Remote I/O error

wx4cb, 2022-02-03:
I zsee the hat in i2cdetect, the screen gets some sort of "snow" pattern on the oled display, and then i get a remote i/o error

any help appreciated

```
charlesb@raspberrypi:~/Pimoroni/automationhat/examples/hat-mini $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

'''
charlesb@raspberrypi:~/Pimoroni/automationhat/examples/hat-mini $ python3 analog.py
analog.py

This Automation HAT Mini example displays the three ADC
analog input voltages numerically and as bar charts.

Press CTRL+C to exit.

Traceback (most recent call last):
  File "/home/charlesb/Pimoroni/automationhat/examples/hat-mini/analog.py", line 70, in <module>
    reading = automationhat.analog[channel].read()
  File "/usr/local/lib/python3.9/dist-packages/automationhat/__init__.py", line 128, in read
    self._update()
  File "/usr/local/lib/python3.9/dist-packages/automationhat/__init__.py", line 132, in _update
    self.setup()
  File "/usr/local/lib/python3.9/dist-packages/automationhat/__init__.py", line 115, in setup
    setup()
  File "/usr/local/lib/python3.9/dist-packages/automationhat/__init__.py", line 376, in setup
    sn3218.enable()
  File "/usr/local/lib/python3.9/dist-packages/sn3218/__init__.py", line 133, in enable
    _get_sn3218().enable()
  File "/usr/local/lib/python3.9/dist-packages/sn3218/__init__.py", line 128, in _get_sn3218
    _sn3218 = SN3218()
  File "/usr/local/lib/python3.9/dist-packages/sn3218/__init__.py", line 34, in __init__
    self.enable_leds(enable_mask)
  File "/usr/local/lib/python3.9/dist-packages/sn3218/__init__.py", line 63, in enable_leds
    self.i2c.write_i2c_block_data(I2C_ADDRESS, CMD_ENABLE_LEDS, [
OSError: [Errno 121] Remote I/O error
charlesb@raspberrypi:~/Pimoroni/automationhat/examples/hat-mini $
'''

Gadgetoid, 2022-02-07:
How's the HAT connected? Pushed down firmly on the header?

Could there be any other software running/starting on boot that's interfering with the GPIO?

wx4cb, 2022-02-07:
the hat is pushed down as far as it will go, nothing is running on there other than the hat (and alll the prerequeaisites re raspi-config has been done as per the instructions.

i've tried it on both a 4pi 4 and pi 3b  on the latest release of raspberry pi os (buster or bullseye not sure which it is)

Gadgetoid, 2022-02-08:
Just fired up on a Pi 4 with the latest 64bit Bullseye and it's working fine, so I suspect it's either a board fault or some kind of complicating factor/configuration issue I'm missing. I've asked the support team to cast their eyes over this and see if they have any ideas and we'll go from there.

Gadgetoid, 2022-02-08:
That's the correct i2c address, I see the same thing:

```
pi@raspberrypi:~/Pimoroni/automationhat/examples/hat-mini $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

Wait, hold on a minute - Automation HAT Mini does not have an SN3218 driver so the error you're getting is very odd indeed. Looks like your board is fine but some software weirdness has happened.

Gadgetoid, 2022-02-08:
GOT IT! Managed to replicate. Working on a fix.

Gadgetoid, 2022-02-08:
Totally my fault, sorry for the run-around here. Fix released. Update with:

```
python3 -m pip install --upgrade automationhat
```

That should get you up and running. :facepalm:

wx4cb, 2022-02-08:
awesome. will take a look at it when i get home today

closed 2022-02-14 by Gadgetoid