pimoroni/automation-hat #57: Unable to use LCD Display on Automation Hat Mini

fussel132, 2024-02-06:
Hi there,

I am not able to get the display to work. Neither the examples here nor the ones over at the st7735 library work. Using the proposed install script (after disabling the externally managed env) still delivers the old examples where 9 instead of "GPIO9" is used as dc parameter.

According to the pinout (found at https://pinout.xyz/pinout/automation_hat_mini), the config below should let me control the display:

`# Create ST7735 LCD display class.
disp = st7735.ST7735(
    port=0,
    cs=st7735.BG_SPI_CS_FRONT,
    dc="GPIO9",
    backlight="GPIO25",
    rotation=90,
    spi_speed_hz=4000000
)`

However it always results in the following output:

> Woah there, suitable gpiochip not found!
  ❌  GPIO9: not found - /dev/gpiochip1 (raspberrypi-exp-gpio)!
  ❌  GPIO9: not found - /dev/gpiochip0 (pinctrl-bcm2711)!

HELP! SPI is enabled via raspi-config. I am running the latest release of headless rpi os 64bit (debian bookworm) as of today on a RPi4 with 4GiB of RAM.

The output of gpioinfo looks like this (I am not an expert but the display isn't there I guess):

<details>
  <summary>gpioinfo output</summary>

  > user@rpi:~ $ gpioinfo
  gpiochip0 - 58 lines:
          line   0:     "ID_SDA"       unused   input  active-high 
          line   1:     "ID_SCL"       unused   input  active-high 
          line   2:       "SDA1"       unused   input  active-high 
          line   3:       "SCL1"       unused   input  active-high 
          line   4:  "GPIO_GCLK"       unused   input  active-high 
          line   5:      "GPIO5"       unused  output  active-high 
          line   6:      "GPIO6"       unused  output  active-high 
          line   7:  "SPI_CE1_N"   "spi0 CS1"  output   active-low [used]
          line   8:  "SPI_CE0_N"   "spi0 CS0"  output   active-low [used]
          line   9:   "SPI_MISO"       unused   input  active-high 
          line  10:   "SPI_MOSI"       unused   input  active-high 
          line  11:   "SPI_SCLK"       unused   input  active-high 
          line  12:     "GPIO12"       unused  output  active-high 
          line  13:     "GPIO13"       unused  output  active-high 
          line  14:       "TXD1"       unused   input  active-high 
          line  15:       "RXD1"       unused   input  active-high 
          line  16:     "GPIO16"       unused  output  active-high 
          line  17:     "GPIO17"       unused   input  active-high 
          line  18:     "GPIO18"       unused   input  active-high 
          line  19:     "GPIO19"       unused   input  active-high 
          line  20:     "GPIO20"       unused   input  active-high 
          line  21:     "GPIO21"       unused   input  active-high 
          line  22:     "GPIO22"       unused   input  active-high 
          line  23:     "GPIO23"       unused   input  active-high 
          line  24:     "GPIO24"       unused   input  active-high 
          line  25:     "GPIO25"       unused  output  active-high 
          line  26:     "GPIO26"       unused   input  active-high 
          line  27:     "GPIO27"       unused   input  active-high 
          line  28: "RGMII_MDIO"       unused   input  active-high 
          line  29:  "RGMIO_MDC"       unused   input  active-high 
          line  30:       "CTS0"       unused   input  active-high 
          line  31:       "RTS0"       unused   input  active-high 
          line  32:       "TXD0"       unused   input  active-high 
          line  33:       "RXD0"       unused   input  active-high 
          line  34:    "SD1_CLK"       unused   input  active-high 
          line  35:    "SD1_CMD"       unused   input  active-high 
          line  36:  "SD1_DATA0"       unused   input  active-high 
          line  37:  "SD1_DATA1"       unused   input  active-high 
          line  38:  "SD1_DATA2"       unused   input  active-high 
          line  39:  "SD1_DATA3"       unused   input  active-high 
          line  40:  "PWM0_MISO"       unused   input  active-high 
          line  41:  "PWM1_MOSI"       unused   input  active-high 
          line  42: "STATUS_LED_G_CLK" "ACT" output active-high [used]
          line  43: "SPIFLASH_CE_N" unused input active-high 
          line  44:       "SDA0"       unused   input  active-high 
          line  45:       "SCL0"       unused   input  active-high 
          line  46: "RGMII_RXCLK" unused input active-high 
          line  47: "RGMII_RXCTL" unused input active-high 
          line  48: "RGMII_RXD0"       unused   input  active-high 
          line  49: "RGMII_RXD1"       unused   input  active-high 
          line  50: "RGMII_RXD2"       unused   input  active-high 
          line  51: "RGMII_RXD3"       unused   input  active-high 
          line  52: "RGMII_TXCLK" unused input active-high 
          line  53: "RGMII_TXCTL" unused input active-high 
          line  54: "RGMII_TXD0"       unused   input  active-high 
          line  55: "RGMII_TXD1"       unused   input  active-high 
          line  56: "RGMII_TXD2"       unused   input  active-high 
          line  57: "RGMII_TXD3"       unused   input  active-high 
  gpiochip1 - 8 lines:
          line   0:      "BT_ON"   "shutdown"  output  active-high [used]
          line   1:      "WL_ON"       unused  output  active-high 
          line   2: "PWR_LED_OFF" "PWR" output active-low [used]
          line   3: "GLOBAL_RESET" unused output active-high 
          line   4: "VDD_SD_IO_SEL" "vdd-sd-io" output active-high [used]
          line   5:   "CAM_GPIO" "cam1_regulator" output active-high [used]
          line   6:  "SD_PWR_ON" "sd_vcc_reg"  output  active-high [used]
          line   7:    "SD_OC_N"       unused   input  active-high 
</details>

Gadgetoid, 2024-02-08:
There are two potential gotchas here:

1. "GPIO9" is named "SPI_MISO" on your board- so you can just swap "GPIO9" to "SPI_MISO" and it... should work
2. "SPI_MISO" *might* be owned by the SPI driver, so it could complain about it being in use

Raspberry Pi are normalising the GPIO names between boards to help avoid things like this (see: https://github.com/raspberrypi/linux/pull/5902)

fussel132, 2024-02-08:
Hm ok so using SPI_MISO indeed helped resolving this issue. Slightly confusing but thanks a lot!

closed 2024-02-08 by fussel132