pimoroni/automation-hat #58: No module named 'gpiodevice'

roger-richards, 2024-02-12:
Using PiZeroW (I know, old but what I had..)
Used following guide to install nodejs on PiZeroW:
[https://hassancorrigan.com/blog/install-nodejs-on-a-raspberry-pi-zero/](https://hassancorrigan.com/blog/install-nodejs-on-a-raspberry-pi-zero/)
Installed Pimoroni software as [documented on their site.](https://learn.pimoroni.com/article/getting-started-with-automation-hat-and-phat)
Enabled I2C via sudo raspi-config

Tried to run output.py and generated error: No module named '**gpiod**'
googled and installed: sudo pip3 install smbus
>same error: No module named '**gpiod**'
googled more and installed: sudo apt install libgpiod-dev python3-libgpiod
>**different** error: No module named '**gpiodevice**'
googled more and installed: sudo pip3 install ST7735
>same error: No module named '**gpiodevice**'

Any suggestions?  I've searched throughly.

roger-richards, 2024-02-12:
duh! had to pip3 install gpiodevice

Gadgetoid, 2024-02-15:
Oof, it might be that Automation HAT is relying on some libraries that have been updated, but which don't properly list their dependencies 😬

I should probably get around to updating Automation HAT itself.

Full details of why everything is a miserable nightmare right now: https://github.com/pimoroni/boilerplate-python/pull/13

And an ongoing progress report: https://github.com/pimoroni/boilerplate-python/issues/16

helgibbons, 2024-05-07:
Hopefully things should be working better now that everything's updated and merged in!

Up-to-date installation instructions can be found [here](https://github.com/pimoroni/automation-hat/) and [here](https://learn.pimoroni.com/article/getting-started-with-automation-hat-and-phat).

Closing this one for now.

closed 2024-05-07 by helgibbons