pimoroni/automation-hat #62: Automation HAT Mini - Source outdated

TheMarrt, 2024-04-27:
HAT Mini was not working after install instructions.

These steps were needed to get the IO and the relay running. I still can't get the display to work

1. pip3 install gpiodevice
2. enable i2c & spi
3. fix some changes in display initialization
https://forums.pimoroni.com/t/issue-with-enviro-with-rpi-zero-display-and-particles-not-working/23365/15

this is how far i got. I removed the display stuff for now so i can at least operate the relay. these are the error messages:

```
Traceback (most recent call last):
  File "/home/ffu/Pimoroni/automationhat/examples/hat-mini/output-mod.py", line 49, in <module>
    disp = st7735.ST7735(
  File "/usr/local/lib/python3.9/dist-packages/st7735/__init__.py", line 175, in __init__
    self._dc = gpiodevice.get_pin(dc, "st7735-dc", OUTL)
  File "/home/ffu/.local/lib/python3.9/site-packages/gpiodevice/__init__.py", line 155, in get_pin
    chip = find_chip_by_pins(pin)
  File "/home/ffu/.local/lib/python3.9/site-packages/gpiodevice/errors.py", line 45, in wrapper
    errors.append(next(i))
  File "/home/ffu/.local/lib/python3.9/site-packages/gpiodevice/__init__.py", line 123, in find_chip_by_pins
    offset = chip.line_offset_from_id(pin_id)
  File "/home/ffu/.local/lib/python3.9/site-packages/gpiod/chip.py", line 144, in line_offset_from_id
    raise ValueError("line offset of out range")
ValueError: line offset of out range
```

helgibbons, 2024-05-07:
Might be worth giving the install another try - we've merged in a bunch of changes to this library over the last week.

Updated installation instructions can be found [here](https://github.com/pimoroni/automation-hat/) and [here](https://learn.pimoroni.com/article/getting-started-with-automation-hat-and-phat).

Let us know if you're still having trouble!

thirdr, 2025-01-09:
Closing this to clear our decks for the new year, if it's still troubling you please re-open! :)